infinition/Bjorn #1: Still Waiting on Commits

Az-Neter, 2024-07-02:
It's been a month. 
Still waiting on literally any commits or updates to this.

Were you lying to everyone?

Collecting stars for a fake project just to rename it later when you fail to follow through?

Jashmane, 2024-07-02:
I would like to know where this project lays? Give us something here?

infinition, 2024-07-02:
Hi,

I've added the README for pre release of the "alpha"
This version will come without OS, only python files.
For the impatients, i'm planning a realease in the coming week (or my best before the end of july).
Please be aware that this version will certainly be full of bugs as i'm alone on this huge project and discovering the joy of dev life ;)
Enjoy the reading before next updates !
Best regards,
Fabien

closed 2024-07-02 by infinition