infinition/Bjorn #5: No RDP.csv file AND display not working

Zandiexoder, 2024-11-07:
Output for RDP.csv file: 
`root@bjorn:~# sudo systemctl status bjorn.service
● bjorn.service - Bjorn Service
     Loaded: loaded (/etc/systemd/system/bjorn.service; enabled; preset: enabled)
     Active: active (running) since Fri 2024-11-08 12:07:02 NZDT; 38s ago
    Process: 889 ExecStartPre=/home/bjorn/Bjorn/kill_port_8000.sh (code=exited, status=0/SUCCESS)
   Main PID: 892 (python3)
      Tasks: 13 (limit: 381)
        CPU: 11.741s
     CGroup: /system.slice/bjorn.service
             └─892 /usr/bin/python3 /home/bjorn/Bjorn/Bjorn.py

Nov 08 12:07:21 bjorn python3[892]:     action_instance = getattr(module, b_class)(self.shared_data)
Nov 08 12:07:21 bjorn python3[892]:                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 08 12:07:21 bjorn python3[892]:   File "/home/bjorn/Bjorn/actions/rdp_connector.py", line 33, in __init__
Nov 08 12:07:21 bjorn python3[892]:     self.rdp_connector = RDPConnector(shared_data)
Nov 08 12:07:21 bjorn python3[892]:                          ^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 08 12:07:21 bjorn python3[892]:   File "/home/bjorn/Bjorn/actions/rdp_connector.py", line 72, in __init__
Nov 08 12:07:21 bjorn python3[892]:     with open(self.rdpfile, "w") as f:
Nov 08 12:07:21 bjorn python3[892]:          ^^^^^^^^^^^^^^^^^^^^^^^
Nov 08 12:07:21 bjorn python3[892]: FileNotFoundError: [Errno 2] No such file or directory: '/home/bjorn/Bjorn/data/output/crackedpwd/rdp.csv'
Nov 08 12:07:35 bjorn python3[892]: 2024-11-08 12:07:35 - display.py - DEBUG - FD : 893`

Unsure why display is not working, but when manually trying to run Bjorn.py is says that GPIO is busy and exits. I think this is due to the script already running though.

Zandiexoder, 2024-11-07:
Display is V4 btw!

Zandiexoder, 2024-11-07:
Update: Missing RSP.csv issue by manually creating the directory /home/bjorn/Bjorn/data/output/crackedpwd using `mkdir /home/bjorn/Bjorn/data/output/crackedpwd -p`. The program then automatically created the required files. 

Display is still not working though!

infinition, 2024-11-07:
Thanks you for your return.
i just fixed it by adding correctly the folders (i forgot the .gitkeep for the empty folders...)

Concerning the blank screen, i have also added a step in the installation (manual & automatic) to choose the screen type.
Sorry for that.

You can create the missing folders manually  :  

mkdir /home/bjorn/Bjorn/data/output
mkdir /home/bjorn/Bjorn/data/logs

and for the screen : sudo vi /home/bjorn/Bjorn/config/shared_config.json
look for the line : "epd_type": "epd2in13_V4",  and change it to "epd_type": "epd2in13_V2"  for example.

save it and restart the service :  sudo systemctl restart bjorn.service

closed 2024-11-07 by infinition