infinition/Bjorn #10: Question

Ademolght, 2024-11-08:
How i enable ssh without see the screen pls ? and how can i know the ip of bjorn ?

Rognils, 2024-11-09:
Put a empty file at the root folder called 'ssh'. You could try to use nmap or perhaps ping bjorn.local.

Or to make life easy, use the "Raspberry Pi Imager" tool, there you can enter your wifi credentials and enable SSH.

RasTacsko, 2024-11-09:
If the wifi credentials were not set up in rpi imager or have to be changed , can we change it after install?

infinition, 2024-11-09:
You can use the web interface, i have created a wifi menu on the config page.

Ademolght, 2024-11-10:
> You can use the web interface, i have created a wifi menu on the config page. @Rognils @RasTacsko @Ademolght

bro thx but i now i can't connect my raspberry pi zero to ssh for install bjorn THIS is the problem

Ademolght, 2024-11-10:
i need format my sd card in FAT32 ?

RasTacsko, 2024-11-10:
If you use raspberry imager it will format and write the data to your SD card

Ademolght, 2024-11-10:
> If you use raspberry imager it will format and write the data to your SD card

i use it

Ademolght, 2024-11-10:
wait in need connect the power port or data port ?

Ademolght, 2024-11-11:
I finaly solved the pb thx @RasTacsko @infinition @Rognils .

closed 2024-11-11 by Ademolght