infinition/Bjorn #7: Waveshare v3

DJPoulter, 2024-11-08:
Waveshare v3 doesn't work due to https://github.com/infinition/Bjorn/blob/4dd30869b3179ca0c7d71de28ced0936792c03f5/resources/waveshare_epd/epd2in13_V3.py#L202

The init function only takes in 1 param compared to the others which take in 2.

It also needs to be added to https://github.com/infinition/Bjorn/blob/4dd30869b3179ca0c7d71de28ced0936792c03f5/shared.py#L173

Else self.screen_reversed is undefined

ZeroCool-Dade, 2024-11-08:
I'm also using v3 and not able to start bjorn service.  my error in logs:

`Nov 08 16:03:00 Bjorn python3[1504]: Error initializing EPD for full update: EPD.init() takes 1 positional argument but 2 were given
Nov 08 16:03:00 Bjorn python3[1504]: 2024-11-08 16:03:00 - shared.py - ERROR - Error initializing EPD display:
Nov 08 16:03:00 Bjorn python3[1504]: EPD.init() takes 1 positional argument but 2 were given
Nov 08 16:03:00 Bjorn python3[1504]: Traceback (most recent call last):
Nov 08 16:03:00 Bjorn python3[1504]:   File "/home/bjorn/Bjorn/Bjorn.py", line 24, in <module>
Nov 08 16:03:00 Bjorn python3[1504]:     from init_shared import shared_data
Nov 08 16:03:00 Bjorn python3[1504]:   File "/home/bjorn/Bjorn/init_shared.py", line 13, in <module>
Nov 08 16:03:00 Bjorn python3[1504]:     shared_data = SharedData()
Nov 08 16:03:00 Bjorn python3[1504]:                   ^^^^^^^^^^^^
Nov 08 16:03:00 Bjorn python3[1504]:   File "/home/bjorn/Bjorn/shared.py", line 38, in __init__
Nov 08 16:03:00 Bjorn python3[1504]:     self.setup_environment() # Setup the environment
Nov 08 16:03:00 Bjorn python3[1504]:     ^^^^^^^^^^^^^^^^^^^^^^^^
Nov 08 16:03:00 Bjorn python3[1504]:   File "/home/bjorn/Bjorn/shared.py", line 170, in setup_environment
Nov 08 16:03:00 Bjorn python3[1504]:     self.initialize_epd_display()
Nov 08 16:03:00 Bjorn python3[1504]:   File "/home/bjorn/Bjorn/shared.py", line 188, in initialize_epd_display
Nov 08 16:03:00 Bjorn python3[1504]:     self.epd_helper.init_full_update()
Nov 08 16:03:00 Bjorn python3[1504]:   File "/home/bjorn/Bjorn/epd_helper.py", line 30, in init_full_update
Nov 08 16:03:00 Bjorn python3[1504]:     self.epd.init(self.epd.lut_full_update)
Nov 08 16:03:00 Bjorn python3[1504]: TypeError: EPD.init() takes 1 positional argument but 2 were given
Nov 08 16:03:02 Bjorn systemd[1]: bjorn.service: Main process exited, code=exited, status=1/FAILURE
Nov 08 16:03:02 Bjorn systemd[1]: bjorn.service: Failed with result 'exit-code'.`

infinition, 2024-11-09:
@DJPoulter @ZeroCool-Dade 
Im trying to create files so you can try and let me know guys, unfortunately i couldnt buy this version when it was on sell 🙏

DJPoulter, 2024-11-10:
I am happy to test when you have the updated files. Else I can also put up an MR this week to fix it.

IncredibleZuess, 2024-11-10:
My MR may fix the issue @DJPoulter #12

ZeroCool-Dade, 2024-11-11:
Confirmed, #22  fixed my 2.13 v3 screen, thanks @jbohack

infinition, 2024-11-12:
Merged the PR, thanks !

closed 2024-11-12 by infinition