infinition/Bjorn #9: Raspberry Pi 4B + WaveShare 3.52 support

Dante9x9, 2024-11-08:
Not sure if this is the right place to put this but hoping for support for Raspberry Pi 4B + WaveShare 3.52. i've attached the installation log from putty
[Bjorn - Pi 4B with 3.52 log.txt](https://github.com/user-attachments/files/17681751/Bjorn.-.Pi.4B.with.3.52.log.txt)

Dante9x9, 2024-11-08:
I've done a bit of poking around but I new to python. Based off what is in resources for epd and shared.py I tried creating an entry for waveshare 3in52 with no luck.

infinition, 2024-11-09:
I need to work of other type of screen👍

neednotapply, 2024-11-18:
Also running a Raspberry Pi 4B here.
Seeing as how Bjorn has a web interface and SSH, I have no intention of using a screen.
(Unless I'm missing something here) The service seems to not have a contingency for headless mode.
It would be much appreciated as it would cut down the cost of unneeded parts.
My logs are very similar to Dante above.