infinition/Bjorn #11: Run bjorn without screen

kurazli, 2024-11-09:
Are there any plans to make this work without a screen attached similar to pwnagotchi and its bluetooth pairing option?
Or am I missing a configuration parameter to make this work already?

infinition, 2024-11-09:
> Are there any plans to make this work without a screen attached similar to pwnagotchi and its bluetooth pairing option?
> 
> Or am I missing a configuration parameter to make this work already?

I am working on this mode 👍

UmutAlihan, 2024-11-11:
looking very forward for björn to be able to run without an ink display; this is currently the only bottleneck I think avoiding this repo to spread in masses :)

however I already doing budget to buy an 2.5 inch screen from aliexpress to build it very much great effort is given here; appreciated 👍

infinition, 2024-11-16:
Already working, multiple users are using it without screen

closed 2024-11-16 by infinition

kurazli, 2024-11-16:
what do i need to set for this?

Sonic0, 2025-01-04:
> what do i need to set for this?

Remove the line "epd_type" from the `/home/bjorn/Bjorn/config/shared_config.json` file.