infinition/Bjorn #13: Question (password word list)

Alex12kip, 2024-11-09:
Hi, i really like this project, but when i was just looking trough the files using a linux pc i found the word list that it uses to crack ssh, but when i tried editing it it said read only, i thought i did something wrong so i reflashed my microsd but i had the same issue. 

Do you have an solución for this? Or is this on purpose?

infinition, 2024-11-09:
> purpose?

How did you edit it ?
sudo vi .... ?
You can simply replace or edit the file with sftp over ssh.

Alex12kip, 2024-11-09:
Yeah i edited it by just plugging it into my linux pc, but ill try sftp over ssh.

Thank you!

infinition, 2024-11-09:
Thanks for your message , let me know 👍

Alex12kip, 2024-11-09:
Yep it works, tysm i couldnt figure out what i was doing wrong, thanks for helping me out

closed 2024-11-12 by infinition