infinition/Bjorn #14: Bjorn's IP address on screen? [Feature request]

Houdini7, 2024-11-10:
Would be helpful to have Bjorn's IP address visible on the screen. 
Especially useful when changing networks via the Config UI and knowing what address to open to monitor via browser.

infinition, 2024-11-11:
Thank you for the suggestion, I will work on a slightly more discreet approach, i think we don’t want to constantly have the IP address on the screen.

Until today, given that the gadget usb was used (pc/ phone / tablet), we didn’t directly need to have the ip of the « target » network.

In addition, we will soon also have the sharing of ip in bluetooth.

What I can do is display the ip in bjorn comment, for 10-15 seconds as soon as he gets the network, I look at possible solutions.

Thanks to you

Houdini7, 2024-11-11:
Thanks for the reply. Great job w/ Bjorn!

Maybe a config option to:  "Display Bjorn IP: [Always | Never | On Boot]" with On Boot the default?

infinition, 2024-11-13:
[image]
Added IP address showing at startup, instead of harcoded on the screen loop

closed 2024-11-13 by infinition

JuanVilla424, 2024-11-13:
#17