infinition/Bjorn #15: Pandas 2.2.3 Hash does not match

sburggsx, 2024-11-10:
Error:

`ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    pandas==2.2.3 from https://www.piwheels.org/simple/pandas/pandas-2.2.3-cp311-cp311-linux_armv6l.whl#sha256=0a30c884638df66782a2f39ab27a2e757f4f398e5ebdc09adcd2ff52d49e4fe8 (from -r requirements.txt (line 6)):
        Expected sha256 0a30c884638df66782a2f39ab27a2e757f4f398e5ebdc09adcd2ff52d49e4fe8
             Got        ba486b00324773401856b9300e2c6ed899e3faccb6a864f69f56dc4379ee00fc
`

pellehardstedt, 2024-11-11:
Hi, @sburggsx @Zakuto3 can you try this?

- Download pip without installing it
`pip download pandas==2.2.3`

- Calculate the hash of the downloaded file 
`sha256sum pandas-2.2.3-cp311-cp311-linux_armv6l.whl`

- take the output and update the pandas line in requirements.txt with the hash like this
`pandas==2.2.3 --hash=sha256:ba486b00324773401856b9300e2c6ed899e3faccb6a864f69f56dc4379ee00fc`

infinition, 2024-11-11:
Hi, what is your os image version ?

Zakuto3, 2024-11-12:
Linux bjorn 6.6.51+rpt-rpi-v6 #1 Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv6l GNU/Linux

beigeworm, 2024-11-12:
I had the same error.. With a different hash on each installation attempt.

`sudo pip install pandas --break-system-packages --no-deps --no-cache-dir`

Running this before installation worked for me.

closed 2024-11-19 by infinition