infinition/Bjorn #16: Port Issues when restarting Bjorn service

IncredibleZuess, 2024-11-10:
# Description
There is an issue with threading where the web server does not exit and unbind the port fully sometimes before the new web server thread is initialized which results in the server ports hopping from 8000 to 8001
# Steps to Reproduce:
Let Bjorn start normally then after a few seconds run:
```bash
sudo systemctl restart bjorn.service
```

# Setup:
### Hardware:
- Raspberry Pi 4B
- 2.7 inch WaveShare E-ink Display V1
### Software:
- Raspberry Pi OS
- System: 32-bit
- Kernel version: 6.6
- Debian version: 12 (bookworm)
# Expected Behaviour:
The Bjorn service restarts and the web server binds to port 8000 successfully

# Actual Behaviour:
## First Stop
[image]
[image]

## Second Stop:
[image]
[image]

# Interesting Note
While writing the issue I noticed that this only happens when the web page is rendered in a browser so it may be that the console being rendered in the web app is adding a few seconds before stopping the server completely

infinition, 2024-11-11:
Yes, unfortunately, i did not manage to free up the 8000 port when i restart the Bjorn service.
This is why i created a .sh file to kill the pid when the service starts.

My only workaround for now was incremental solution.But i hope this problem will be solved in next releases

infinition, 2024-11-25:
Solved in the upcoming release.

closed 2024-11-25 by infinition