infinition/Bjorn #23: Web UI Stops Working

jaxpr3394, 2024-11-11:
After ~30 minutes of the web UI working, it will stop, and Bjorn will also stop updating the e-ink display. Monitoring the logs, I see a python error:

`ERROR - Unexpected error: [Errno 24] Too many open files: '/home/bjorn/Bjorn/web/screen.png'`

The workaround here is to stop and start the Bjorn service every ~30 minutes. I noticed in both `display.py` and `utils.py`, you are opening `screen.png`, but I never see a `close()` call. I added that call to both of those files, but it did not alleviate the issue, so I figured I would open an issue to get your thoughts.

jaxpr3394, 2024-11-11:
Looks like the latest error in the web UI was:

`{"status": "error", "message": "[Errno 24] Too many open files: '/home/bjorn/Bjorn/data/logs/temp_log.txt'"}`

And `cat temp_log.txt` gives the the latest entries as:

```
==> /home/bjorn/Bjorn/data/logs/webapp.py.log <==
2024-11-11 22:44:33 - webapp.py - ERROR - Unexpected error: [Errno 24] Too many open files: '/home/bjorn/Bjorn/web/screen.png'
```

infinition, 2024-11-11:
Yes, i am aware about this, unfortunately.
On README i said i was trying to solve File descriptor limit.

I literrally spent weeks tweaking everything to solve this error.

Can you tell me  :

sudo vi /etc/systemd/system.conf
and 
sudo vi /etc/systemd/user.conf

DefaultLimitNOFILE=65535  should be uncommented.

And can you ensure to have these 6 lines at the end of :
sudo vi /etc/security/limits.conf

# End of file
root soft nofile 65535
root hard nofile 65535
bjorn soft nofile 65535
bjorn hard nofile 65535
* soft nofile 65535
* hard nofile 65535

Then 

sudo sysctl -p
sudo systemctl daemon-reload
sudo reboot

Thanks

jbohack, 2024-11-11:
@infinition I noticed this issue as well and did confirm those limits were applied with `ulimit -n`. It looks like this may be related to an error within `scan_with_semaphore` that causes resource exhaustion after erroring. I've been trying to debug this, without luck so far. However, feel free to use this as a reference to hopefully help resolve the issue. I will continue to look as well.

[image]

infinition, 2024-11-12:
Thanks for your feedback.

I had a strange value, due to the autoinstall script, can you confirme me that you have this line under :

sudo vi /etc/systemd/system.conf

```
DefaultLimitNOFILE=65535  #should be uncommented.
```

Concerning the  scan_with_semaphore , the thing is i completely tried to run Bjorn without the scanner (with scan_with_semaphore  thing) and i had this file descriptor limit error.
I think it may be related to the fact that i use screenshot of bjorn screen to display it live(with a delay) on the webpage .
I keep working hard on it.

jbohack, 2024-11-12:
> Thanks for your feedback.
>
> I had a strange value, due to the autoinstall script, can you confirme me that you have this line under :
>
> sudo vi /etc/systemd/system.conf
>
> ```
> DefaultLimitNOFILE=65535 [...]

Mine is currently set to `DefaultLimitNOFILE=655351024:524288`. Sounds like that thread issue may just be a side-effect of out of resources if that is the case.

jaxpr3394, 2024-11-12:
Just catching up here. I implemented the proposed solution and it seems to be fine so far. I have had it running for ~45m, which is longer than it generally runs. I will update you again in a bit.

infinition, 2024-11-12:
#27  #

closed 2024-11-12 by infinition