infinition/Bjorn #26: Add static assignment of IP

LoveMeAShitshow, 2024-11-12:
So there's a lot of devices on my network which occasionally leads to DHCP misasigning IPs and setting ip via ssh ifconfig wlan0 [local ip address] doesn't change what orchestrator scans even once I have turned it off and back on. Is there any plan to add static ip assignment to the project in the roadmap?

infinition, 2024-11-12:
Hi ! I'm not entirely sure I fully understand your issue, but I wanted to clarify that, for now, Bjorn is just a .sh auto-installation script. I've left the OS and network configuration up to each user to handle individually.

That said, static IP configuration may indeed become a subject once we release a version with a complete OS image.

As for the orchestrator, it doesn’t perform the scan itself. Instead, it calls the scanning.py script, which retrieves the subnet from the local IP address to scan that specific subnet.

LoveMeAShitshow, 2024-11-12:
My home network runs on 192.168.1.0/24. When connecting to my network it's assigning to a 192
168.20.0/24 address, it's been a problem with my network, when I change the IP to static using ifconfig it doesn't change the subnet of the ip and upon restarting it doesn't save the static ip for the network.

LoveMeAShitshow, 2024-11-12:
I found a workaround using nmcli instead of ifconfig

closed 2024-11-25 by infinition