infinition/Bjorn #35: Enhancement Request: Add Interface Selection for Wi-Fi Scanning

prokyle123, 2024-11-13:
* Description**  
The current plugin scans any Wi-Fi adapter that has an IP but gets stuck on the first one it finds, often leading it to select the wrong interface. In my setup, I use a Wi-Fi USB adapter on `wlan1` as an SSH backbone to maintain a constant connection and leave `wlan0` for scanning. However, the plugin keeps scanning `wlan1`, preventing `wlan0` from being used for its intended purpose.

**Proposed Enhancement**  
Enhance the plugin to allow explicit interface selection or improve its logic to avoid adapters with active IP connections, ensuring that the correct interface is used for scanning.

**Suggested Solutions**  
1. **Interface Selection Option**  
   - Add a parameter (e.g., `--interface wlan0`) or a configuration setting to specify which interface to use for scanning.
  

Interface Exclusion Option**  
   - Add a feature to exclude specific interfaces (e.g., `wlan1`) from scanning. This would focus the plugin on `wlan0`, ensuring it performs as intended.

**Impact**  
This enhancement would provide greater flexibility for setups using multiple adapters, ensuring that dedicated network roles are respected.

**Environment**  
- Plugin Version: [MainBranch
- System: [As Required]
- Network Setup:
  - `wlan1`: Used for SSH backbone connection
  - `wlan0`: Reserved for Wi-Fi scanning

**Additional Notes**  
Please let me know if further details or logs are needed. Thank you for considering this improvement!

infinition, 2024-11-19:
I will try to fix it thanks for the feedback

prokyle123, 2024-11-20:
Thank you! if you need any help im happy to help.

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.

DUptain1993, 2026-09-12:
Addressed in https://github.com/infinition/Bjorn/pull/193 via optional `scan_interface` config (set in Settings / shared_config.json to e.g. `wlan0` or `wlan1`).