infinition/Bjorn #40: Orchestrator don't find WiFi

Mitchou, 2024-11-16:
**Describe the bug**
On a fresh install (32Bits or 64Bits), everything work fine except that Orcherstrator didn't find WiFi
**To Reproduce**
Steps to reproduce the behavior:

1. Fresh install (script or manual)'
2. **Change default local (fr_FR.UTF-8 for example)**
3. Connect to web interface and see logs

**Expected behavior**
Working network scan

**Screenshots**

**Hardware (please complete the following information):**

- Device: Rpi 2W
- OS: Raspbian 32bits or 64bits
- Browser : Firefox
- Version: 130.0.1

**Fix**
If I change change default local back to en_GB.UTF-8, it's works.
So maybe add to the documentation to not change local, or change code of Orchestrator :)

meditant, 2024-11-19:
Same problems and same solution !

AllevatoreAnonimo, 2025-06-27:
But why does this happen?

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.