infinition/Bjorn #44: Service fails to start

afreeland, 2024-11-18:
**Describe the bug**
After switching from one WiFi network to another, it appeared to work for a little while and then stopped working.  

```
Nov 17 21:11:32 bjorn systemd[1]: bjorn.service: Scheduled restart job, restart counter is at 8.
Nov 17 21:11:32 bjorn systemd[1]: Stopped bjorn.service - Bjorn Service.
Nov 17 21:11:32 bjorn systemd[1]: bjorn.service: Consumed 1.492s CPU time.
Nov 17 21:11:32 bjorn systemd[1]: Starting bjorn.service - Bjorn Service...
Nov 17 21:11:32 bjorn systemd[1]: Started bjorn.service - Bjorn Service.
Nov 17 21:11:33 bjorn python3[833]: 2024-11-17 21:11:33 - shared.py - INFO - Loading configuration...
Nov 17 21:11:33 bjorn python3[833]: Traceback (most recent call last):
Nov 17 21:11:33 bjorn python3[833]:   File "/home/bjorn/Bjorn/Bjorn.py", line 24, in <module>
Nov 17 21:11:33 bjorn python3[833]:     from init_shared import shared_data
Nov 17 21:11:33 bjorn python3[833]:   File "/home/bjorn/Bjorn/init_shared.py", line 13, in <module>
Nov 17 21:11:33 bjorn python3[833]:     shared_data = SharedData()
Nov 17 21:11:33 bjorn python3[833]:                   ^^^^^^^^^^^^
Nov 17 21:11:33 bjorn python3[833]:   File "/home/bjorn/Bjorn/shared.py", line 43, in __init__
Nov 17 21:11:33 bjorn python3[833]:     self.update_mac_blacklist()
Nov 17 21:11:33 bjorn python3[833]:   File "/home/bjorn/Bjorn/shared.py", line 176, in update_mac_blacklist
Nov 17 21:11:33 bjorn python3[833]:     if mac_address not in self.config['mac_scan_blacklist']:
Nov 17 21:11:33 bjorn python3[833]:        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 17 21:11:33 bjorn python3[833]: TypeError: argument of type 'NoneType' is not iterable
Nov 17 21:11:33 bjorn systemd[1]: bjorn.service: Main process exited, code=exited, status=1/FAILURE
Nov 17 21:11:33 bjorn systemd[1]: bjorn.service: Failed with result 'exit-code'.
Nov 17 21:11:33 bjorn systemd[1]: bjorn.service: Consumed 1.477s CPU time.
Nov 17 21:11:33 bjorn systemd[1]: bjorn.service: Scheduled restart job, restart counter is at 9.
Nov 17 21:11:33 bjorn systemd[1]: Stopped bjorn.service - Bjorn Service.
Nov 17 21:11:33 bjorn systemd[1]: bjorn.service: Consumed 1.477s CPU time.
Nov 17 21:11:33 bjorn systemd[1]: bjorn.service: Start request repeated too quickly.
Nov 17 21:11:33 bjorn systemd[1]: bjorn.service: Failed with result 'exit-code'.
Nov 17 21:11:33 bjorn systemd[1]: Failed to start bjorn.service - Bjorn Service.
```

**To Reproduce**
I'm not entirely sure but perhaps changing from one WiFi network to another via the web interface.

**Expected behavior**
Service should be able to restart

**Screenshots**
n/a

**Hardware (please complete the following information):**

- Device: Pi Zero 2 W (Waveshare 2.13 v4)
- OS: raspi lite 64 bit
- Browser n/a
- Version n/a

**Additional context**
Looking further up in the logs in looks like this might be the source.  These 192.168.40.0/24 network is when I changed it, not sure if its related or not...

It does look like perhaps files don't get properly close?
> File descriptor threshold reached: 64858 (threshold: 64535). Restarting service

```
Nov 17 13:38:11 bjorn python3[371781]: targets. Next scan in: 0 seconds
Nov 17 13:38:12 bjorn python3[371781]: 2024-11-17 13:38:12 - display.py - DEBUG - FD : 12587
Nov 17 13:38:12 bjorn python3[371781]: 2024-11-17 13:38:12 - shared.py - INFO - Initializing the network knowledge base
Nov 17 13:38:12 bjorn python3[371781]: CSV file with headers
Nov 17 13:38:12 bjorn python3[371781]: 2024-11-17 13:38:12 - shared.py - INFO - Network knowledge base CSV file already
Nov 17 13:38:12 bjorn python3[371781]: exists at /home/bjorn/Bjorn/data/netkb.csv
Nov 17 13:38:12 bjorn python3[371781]: 2024-11-17 13:38:12 - orchestrator.py - INFO - Executing action SSHBruteforce
Nov 17 13:38:12 bjorn python3[371781]: for 192.168.40.65:22
Nov 17 13:38:12 bjorn python3[371781]: 2024-11-17 13:38:12 - ssh_connector.py - INFO - Executing SSHBruteforce on
Nov 17 13:38:12 bjorn python3[371781]: 192.168.40.65:22...
Nov 17 13:38:12 bjorn python3[371781]: 2024-11-17 13:38:12 - ssh_connector.py - INFO - Running bruteforce_ssh on
Nov 17 13:38:12 bjorn python3[371781]: 192.168.40.65:22...
Nov 17 13:38:20 bjorn python3[371781]:   Bruteforcing SSH... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100%
Nov 17 13:38:21 bjorn python3[371781]: 2024-11-17 13:38:21 - orchestrator.py - INFO - Executing action SSHBruteforce
[... 120 lines cut ...]
Nov 17 13:39:33 bjorn python3[371781]:     handle_exit_display(sig, frame, display_thread)
Nov 17 13:39:33 bjorn python3[371781]:   File "/home/bjorn/Bjorn/display.py", line 373, in handle_exit_display
Nov 17 13:39:33 bjorn python3[371781]:     sys.exit(0)
Nov 17 13:39:33 bjorn python3[371781]: SystemExit: 0
Nov 17 13:39:36 bjorn systemd[1]: bjorn.service: Deactivated successfully.
Nov 17 13:39:36 bjorn systemd[1]: Stopped bjorn.service - Bjorn Service.
Nov 17 13:39:36 bjorn systemd[1]: bjorn.service: Consumed 26min 45.253s CPU time.
Nov 17 13:39:36 bjorn systemd[1]: Starting bjorn.service - Bjorn Service...
Nov 17 13:39:36 bjorn systemd[1]: Started bjorn.service - Bjorn Service.
Nov 17 13:39:37 bjorn python3[399946]: 2024-11-17 13:39:37 - shared.py - INFO - Loading configuration...
Nov 17 13:39:37 bjorn python3[399946]: Traceback (most recent call last):
Nov 17 13:39:37 bjorn python3[399946]:   File "/home/bjorn/Bjorn/Bjorn.py", line 24, in <module>
Nov 17 13:39:37 bjorn python3[399946]:     from init_shared import shared_data
Nov 17 13:39:37 bjorn python3[399946]:   File "/home/bjorn/Bjorn/init_shared.py", line 13, in <module>
Nov 17 13:39:37 bjorn python3[399946]:     shared_data = SharedData()
Nov 17 13:39:37 bjorn python3[399946]:                   ^^^^^^^^^^^^
Nov 17 13:39:37 bjorn python3[399946]:   File "/home/bjorn/Bjorn/shared.py", line 43, in __init__
Nov 17 13:39:37 bjorn python3[399946]:     self.update_mac_blacklist()
Nov 17 13:39:37 bjorn python3[399946]:   File "/home/bjorn/Bjorn/shared.py", line 176, in update_mac_blacklist
Nov 17 13:39:37 bjorn python3[399946]:     if mac_address not in self.config['mac_scan_blacklist']:
Nov 17 13:39:37 bjorn python3[399946]:        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 17 13:39:37 bjorn python3[399946]: TypeError: argument of type 'NoneType' is not iterable
Nov 17 13:39:38 bjorn systemd[1]: bjorn.service: Main process exited, code=exited, status=1/FAILURE
Nov 17 13:39:38 bjorn systemd[1]: bjorn.service: Failed with result 'exit-code'.
```

infinition, 2024-11-18:
Try to delete the blacklisted mac adress in the file Bjorn\config\shared_config.json  and restart the service or the rpi

Concerning the second message :

Nov 17 13:39:31 bjorn bash[371784]: File descriptor threshold reached: 64858 (threshold: 64535). Restarting service.
This is normal as we are limited with the FD descriptor , we created a service to restart bjon once it's reached.
For how many hours was Bjorn launched till the service restart ?

afreeland, 2024-11-18:
I did update 
```
"mac_scan_blacklist": null,
"ip_scan_blacklist": null,
```
to 
```
  "mac_scan_blacklist": [],
  "ip_scan_blacklist": [],
  ```
  
  And it looks the service is back up and running

afreeland, 2024-11-18:
Looked into it real quick and determined it to be from saving from the web UI > config.js.  I created a PR here: https://github.com/infinition/Bjorn/pull/49

afreeland, 2024-11-19:
Closing this issue since it has been resolved by this PR:
https://github.com/infinition/Bjorn/pull/49

afreeland, 2024-11-19:
Closing this issue since it has been resolved by: 
https://github.com/infinition/Bjorn/pull/49

closed 2024-11-19 by afreeland