infinition/Bjorn #50: bjorn.service runs twice

afreeland, 2024-11-19:
**Describe the bug**
It looks like the `bjorn.service` runs twice?

I noticed while looking into some FileDescriptor issue that the `bjorn.service` appears to be running two times, event after stopping and starting it back up.

```
bjorn@bjorn:~ $ ps aux | grep bjorn.service
root       31507  0.0  0.3   6800  1684 ?        S    19:36   0:00 /bin/bash -c FILE_LIMIT=$(ulimit -n); THRESHOLD=$(( FILE_LIMIT - 1000 )); while :; do TOTAL_OPEN_FILES=$(lsof | wc -l); if [ "$TOTAL_OPEN_FILES" -ge "$THRESHOLD" ]; then echo "File descriptor threshold reached: $TOTAL_OPEN_FILES (threshold: $THRESHOLD). Restarting service."; systemctl restart 
bjorn.service; exit 0; fi; sleep 10; done &
root       42949  0.0  0.2   6800  1280 ?        S    20:55   0:00 /bin/bash -c FILE_LIMIT=$(ulimit -n); THRESHOLD=$(( FILE_LIMIT - 1000 )); while :; do TOTAL_OPEN_FILES=$(lsof | wc -l); if [ "$TOTAL_OPEN_FILES" -ge "$THRESHOLD" ]; then echo "File descriptor threshold reached: $TOTAL_OPEN_FILES (threshold: $THRESHOLD). Restarting service."; systemctl restart 
bjorn.service; exit 0; fi; sleep 10; done &
bjorn      42976  0.0  0.4   6224  1920 pts/0    S+   20:56   0:00 grep --color=auto bjorn.service

bjorn@bjorn:~ $ sudo systemctl stop bjorn.service

bjorn@bjorn:~ $ ps aux | grep bjorn.service
bjorn      42986  0.0  0.4   6224  1920 pts/0    S+   20:56   0:00 grep --color=auto bjorn.service

bjorn@bjorn:~ $ sudo systemctl start bjorn.service

bjorn@bjorn:~ $ ps aux | grep bjorn.service
root       42996  0.0  0.3   6800  1556 ?        S    20:56   0:00 /bin/bash -c FILE_LIMIT=$(ulimit -n); THRESHOLD=$(( FILE_LIMIT - 1000 )); while :; do TOTAL_OPEN_FILES=$(lsof | wc -l); if [ "$TOTAL_OPEN_FILES" -ge "$THRESHOLD" ]; then echo "File descriptor threshold reached: $TOTAL_OPEN_FILES (threshold: $THRESHOLD). Restarting service."; systemctl restart bjorn.service; exit 0; fi; sleep 10; done &
root       42997  0.0  0.3   6800  1304 ?        S    20:56   0:00 /bin/bash -c FILE_LIMIT=$(ulimit -n); THRESHOLD=$(( FILE_LIMIT - 1000 )); while :; do TOTAL_OPEN_FILES=$(lsof | wc -l); if [ "$TOTAL_OPEN_FILES" -ge "$THRESHOLD" ]; then echo "File descriptor threshold reached: $TOTAL_OPEN_FILES (threshold: $THRESHOLD). Restarting service."; systemctl restart bjorn.service; exit 0; fi; sleep 10; done &
bjorn      43008  0.0  0.4   6224  1920 pts/0    S+   20:56   0:00 grep --color=auto bjorn.service
```

**To Reproduce**

**Expected behavior**
It should only run a single instance

**Screenshots**
[image]

**Hardware (please complete the following information):**

- Device: pi zero 2 w
- OS: raspbian lite 64 bit

**Additional context**
Add any other context about the problem here.

leow149, 2024-11-19:
This may be a different issue, but when I run an IP-scan while Bjorn is running, I can see that it also gets 2 IP Addresses.
Maybe that's related.

infinition, 2024-11-19:
Really strange , have you tried to re imaged ?

Houdini7, 2024-11-19:
Did you happen to 'Install' Bjorn multiple times?
I saw the same thing but was after I ran the install script multiple times...

infinition, 2024-11-25:
Indeed, it's seems to happens when the script have been ran multiple time.
Reopen that case if not solved with a fresh install on a new OS.

closed 2024-11-25 by infinition

leow149, 2024-11-25:
I just flashed a fresh image and ran the install script. Still the same outcome.

[image]