infinition/Bjorn #64: Stuck at awakening.

JustEvanB, 2024-11-28:
# Bjorn is stuck at "IDLE Awakening"

**Why**
Steps to reproduce the behavior:

Replaced "passwords.txt" with the rockyou.txt wordlist.
Converted the rockyou to ASCII and changed permissions to match with old password.txt
booting into IDLE

**Screenshots**
[image]

also stopped bjorn.service and run:
```
sudo python ./Bjorn.py
```
[image]

then here is the new and default wordlists, as shown the new passwords.txt are the exact same.
with passwords.txt.BKP being the default password list that comes with Bjorn.
[image]

JuanVilla424, 2024-11-30:
Greetings. I think u catched a soft limit on rpi 0.

Thats a knowledge behaivour.  rocku.txt has 13M of passwords, that cause your rpi 0 no able to process all list cuz even doesnt have sufficient memory to load it.  

I ll try to solve it using a RPI 4/5 and SSD. I ll be back with more info.

JustEvanB, 2024-12-03:
Thank you for the response, I will cut down rockyou down, and mark this as closed.

closed 2024-12-03 by JustEvanB