infinition/Bjorn #68: usb0 ip not assigned

Stinocon, 2024-12-04:
I am trying to connect to the local usb0 interface so that I can check bjorn and possibly change WiFi networks to scan.
Unfortunately, however, I can only ever reach it through the IP of the WiFi network I initially connected it to. 

I use a Mac and if I connect bjorn via USB I see it properly listed in the network interfaces. Clearly I set my interface as listed on 172.20.2.2 etc etc.
Via SSH by running an _ip addr_ I see that the usb0 interface (which should have 172.20.2.1) has an apipa address. 

The configuration in /etc/network/interfaces seems correct

_allow-hotplug usb0
iface usb0 inet static
    address 172.20.2.1
    netmask 255.255.255.0_

Suggestions? Thanks

_2: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff
    inet 169.254.98.1/16 brd 169.254.255.255 scope global noprefixroute usb0
       valid_lft forever preferred_lft forever
    inet6 fe80::1caa:21ed:27f4:9bce/64 scope link 
       valid_lft forever preferred_lft forever_

timewall, 2024-12-05:
Look in the file /etc/networks, your IP is there

The two settings seem to be interfering with each other.

At least Björn can be reached at IP 169.254.98.1. I haven't found a quick solution.

Stinocon, 2024-12-06:
Okay... I was able to create a workaround.
I took /etc/networks and changed the link-local subnet to reflect what it should be:

172.20.2.0

Then I edited /etc/dhcpcd.conf by adding at the end of the file

interface usb0
static ip_address=172.20.2.1/24

After rebooting, the device responds correctly on address 172.20.2.1

michaelbapst, 2025-01-19:
This fixed everything! Thank you!

HanSyt, 2025-06-01:
I think this  behavior is caused by NetworkManager, that replaces the more native methods.
I have a different approach (yeah it's Linux and you can), to leave the apipa config in networks not touched.

Instead I manually add the IP address at boot through crontab.

sudo crontab -e (it is supposed to run as root) and add this:
@reboot `/usr/sbin/ip a add 172.20.2.1/24 dev usb0`

After a reboot you will see with ip a if the action has succeeded. 

This also give you the opportunity to login through ssh if the default network connection is not available.
Add a new network with nmtui command, which gives you a nice interface to add a new Wifi connection.

biazzotto, 2025-06-19:
@HanSyt, with "be aware the \` at the beginning and end of the command", do you mean?
> @reboot \`/usr/sbin/ip a add 172.20.2.1/24 dev usb0\`

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.