infinition/Bjorn #72: Waveshare 2.13 rev 2.1 epd2in13g

sqvist, 2024-12-06:
I connected an e-ink display to the gpio-pins. And nothing happened eventhough i tried all epd-settings in the config.json.

So i downloaded demos from Waveshare, and it seems that my version of display is "epd2in13g". 
This because their python-example epd_2in13g_test.py makes my screen function (noteworthy, with 8-pin connector).

It doesnt seem that bjorn works with this "g"-type, am i doing something wrong, or is it just not supported at the moment?

Im testing on a pi zero 2 w.

megastruktur, 2024-12-13:
I have a rev 2.1 with Pi Zero 2W.
Looks like 2.1 is a V3. So I've selected "v3" while installing. In addition to that, before the first reboot right after the installation, I've enabled SPI and I2C though the raspi-config. Then it worked like a charm.

Note: My rev 2.1 is connected to Pi Zero 2W via GPIO mount.

Davelicious, 2025-02-23:
It still doesn't work on my rev 2.1 with Pi Zero 2W
also the web interface isn't working

DUptain1993, 2026-09-12:
`epd2in13g` is a different Waveshare driver family than the panels added in https://github.com/infinition/Bjorn/pull/193 (`epd2in13b_v3`). Keeping open as a hardware-support request.