infinition/Bjorn #81: Error when manually selecting attacks from web interface

Terrib1e, 2024-12-15:
**Describe the bug**
When going to the manual attack section in the web interface, any selection gives the following error:

Error code: 404        
Message: File not found.
Error code explanation: 404 - Nothing matches the given URI."

Talismancer, 2024-12-27:
Same...had hoped for a log output of the attack. 
In the Browser developer mode I get the following:

```
index.html:106  POST http://10.0.0.2:8000/execute_manual_attack 500 (Internal Server Error)
index.html:118 Failed to execute manual attack: Action class NetworkScanner not found
index.html:126 GET http://10.0.0.2:8000/recent_logs 404 (File not found)

```
(I'm using a USB connection using the same IP  as I use for Pwnagotchi to avoid changing Network Connection properties everytime I switch microSD cards)

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.