infinition/Bjorn #91: No module named 'rich'

PeterHurn, 2025-01-02:
Bjorn fails to start on Raspberry Pi W v1.1 with epaper hat with minimal install of bookworm OS

Part way through full install I receive the following:

Collecting pandas==2.2.3
  Downloading https://www.piwheels.org/simple/pandas/pandas-2.2.3-cp311-cp311-linux_armv6l.whl (43.2 MB)
     ━━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━━━━━ 17.6/43.2 MB 576.4 kB/s eta 0:00:45
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    pandas==2.2.3 from https://www.piwheels.org/simple/pandas/pandas-2.2.3-cp311-cp311-linux_armv6l.whl#sha256=0a30c884638df66782a2f39ab27a2e757f4f398e5ebdc09adcd2ff52d49e4fe8 (from -r requirements.txt (line 6)):
        Expected sha256 0a30c884638df66782a2f39ab27a2e757f4f398e5ebdc09adcd2ff52d49e4fe8
             Got        d58e0fea5e324e0684efa40e4102ccc145d27ba757a7e34de454f80c05a52681

[2025-01-02 16:17:46] [ERROR] An error occurred during: Installed Python requirements (Error code: 1)
[2025-01-02 16:17:46] [ERROR] Check the log file for details: /var/log/bjorn_install/bjorn_install_20250102_160759.log

Would you like to:
1. Retry this step
2. Skip this step (not recommended)
3. Exit installation\033[0m

I selected 1 and it seemed to continue but since I see the error above it's probably skipping some of the installation and thus failing.

Running  journalctl -fu bjorn.service shows the following repeatidly:

Jan 02 16:24:37 bjorn python3[1572]:     from rich.console import Console
Jan 02 16:24:37 bjorn python3[1572]: ModuleNotFoundError: No module named 'rich'
Jan 02 16:24:37 bjorn systemd[1]: bjorn.service: Main process exited, code=exited, status=1/FAILURE
Jan 02 16:24:37 bjorn systemd[1]: bjorn.service: Failed with result 'exit-code'.
Jan 02 16:24:38 bjorn systemd[1]: bjorn.service: Consumed 2.285s CPU time.
Jan 02 16:24:38 bjorn systemd[1]: bjorn.service: Scheduled restart job, restart counter is at 93.
Jan 02 16:24:38 bjorn systemd[1]: Stopped bjorn.service - Bjorn Service.
Jan 02 16:24:38 bjorn systemd[1]: bjorn.service: Consumed 2.285s CPU time.
Jan 02 16:24:38 bjorn systemd[1]: Starting bjorn.service - Bjorn Service...
Jan 02 16:24:38 bjorn systemd[1]: Started bjorn.service - Bjorn Service.
Jan 02 16:24:40 bjorn python3[1586]: Traceback (most recent call last):
Jan 02 16:24:40 bjorn python3[1586]:   File "/home/bjorn/Bjorn/Bjorn.py", line 24, in <module>
Jan 02 16:24:40 bjorn python3[1586]:     from init_shared import shared_data
Jan 02 16:24:40 bjorn python3[1586]:   File "/home/bjorn/Bjorn/init_shared.py", line 11, in <module>
Jan 02 16:24:40 bjorn python3[1586]:     from shared import SharedData
Jan 02 16:24:40 bjorn python3[1586]:   File "/home/bjorn/Bjorn/shared.py", line 25, in <module>
Jan 02 16:24:40 bjorn python3[1586]:     from logger import Logger
Jan 02 16:24:40 bjorn python3[1586]:   File "/home/bjorn/Bjorn/logger.py", line 20, in <module>
Jan 02 16:24:40 bjorn python3[1586]:     from rich.console import Console
Jan 02 16:24:40 bjorn python3[1586]: ModuleNotFoundError: No module named 'rich'

ohsnaplaszlo, 2025-01-06:
I was having the same issue but just reran the install script a couple of times and it resolved itself.  Not sure if it's helpful or not but, it did get resolved.

Each time it happened the download was at a different percent.  My guess is that the download was failing and the package (and subsequent packages) wasn't getting installed.

Davelicious, 2025-03-04:
I have issue the on my Raspberry Pi W2 
The above solution doesn't work the display stays empty.
I tried it with the advised & latest 32 & 64 bit RB lite versions
and with 2 different e-paper Hat displays v3 (rev v2.1) & a v4 :(

What else can I do to get bjorn working when the instructions don't work?

Rotzke, 2025-05-10:
@Davelicious just fixed my problem with rich and pandas by installing them using `sudo apt install python3-rich python3-pandas`, for some time you can't install pip libraries globally in Debian.

magiker, 2025-05-11:
> Rotzke

Stop posting.

Not using that email

Davelicious, 2025-05-16:
@Rotzke thx this got me a little bit further, the display is finally working.

Rotzke, 2025-05-16:
@Davelicious I still had to restart the script 3 more times because not all pip dependencies were installed correctly. I guess it is because of Raspberry Pi Zero capabilities. I had multiple retries on the endpoint of that additional pip registry. So, I guess it would be a great idea to run the script a couple of times more on your side.

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.