infinition/Bjorn #107: Error when using quick install sh.

popeyesfx, 2025-03-11:
**Describe the bug**
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    pandas==2.2.3 from https://www.piwheels.org/simple/pandas/pandas-2.2.3-cp311-cp311-linux_armv7l.whl#sha256=0a30c884638df66782a2f39ab27a2e757f4f398e5ebdc09adcd2ff52d49e4fe8 (from -r requirements.txt (line 6)):
        Expected sha256 0a30c884638df66782a2f39ab27a2e757f4f398e5ebdc09adcd2ff52d49e4fe8
             Got        10b34c28e00d32f418a47946a4c5bcf337fd6fa823eac5841227e9c4c1e49e63

[2025-03-11 19:43:49] [ERROR] An error occurred during: Installed Python requirements (Error code: 1)
[2025-03-11 19:43:49] [ERROR] Check the log file for details: /var/log/bjorn_install/bjorn_install_20250311_175121.log

Would you like to:
1. Retry this step
2. Skip this step (not recommended)
3. Exit installation\033[0m

**To Reproduce**
Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
No error message 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**

- Device: Pi zero 2 w
- OS: PI OS bookworm lite 32bit 2024-11-19
- Browser safari
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

popeyesfx, 2025-03-11:
Also used the recommended image and still get the same error

popeyesfx, 2025-03-12:
After several hrs trying to work out what was going on with no luck. Decided to re-download the image and start again with a brand new sd card. This did the trick. I can only surmise that one or both were faulty.

renegade2k, 2025-03-13:
Would suggest you close the issue then ...

pyrophreak, 2025-09-17:
> After several hrs trying to work out what was going on with no luck. Decided to re-download the image and start again with a brand new sd card. This did the trick. I can only surmise that one or both were faulty.

I tried the same but using a different card, screen and pi zero board I continue to get the same error. Is there a command to run the script to bypass or turn off hash checking so it will just install correctly. or what file do we need to open to modify/update the hashes?

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.