infinition/Bjorn #111: nmcli taking 100% of a core

MagicalSpacePope, 2025-04-28:
**Describe the bug**
The command "nmcli -t -f active dev wifi" is consuming 100% of a core shortly after boot. 

**To Reproduce**
Steps to reproduce the behavior: 

1. Reboot
2. Check utilization, "top, htop, btop, ps aux, etc"
3. Observe utilization

**Expected behavior**
If this doesnt need to run constantly can the process be changed?

**Screenshots**

[image]

**Hardware (please complete the following information):**

- Device: Rpi Zero 2
- OS: Raspbian Lite

HanSyt, 2025-05-21:
I noticed this too

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.