infinition/Bjorn #113: Waveshare v4: unreadable

00ihsan, 2025-05-20:
**Describe the bug**
Screen not readable.

**To Reproduce**
Waveshare v4 display on raspberry pi 2w with debian.

**Expected behavior**
A clear display that is readable.

**Screenshots**
I'll add as a comment

**Hardware (please complete the following information):**

- RPi 0W2
- Debian

00ihsan, 2025-05-20:
[image]

00ihsan, 2025-05-20:
[image]

HanSyt, 2025-05-21:
Same here, however in the config file I changed V4 to V3 and it is readable now.

00ihsan, 2025-05-21:
Thanks but did not do the trick for me.

christdoesselaere, 2025-09-14:
> Same here, however in the config file I changed V4 to V3 and it is readable now.

Can confirm. Had the same issue, this fixed it.

tacoroumen, 2025-10-25:
switchin from v4 to v3 and then rebooting also fixed it for me took a long time to boot after the change tho (it might be because i have the 3 colour v4) because i see some red details now in the display

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.

DUptain1993, 2026-09-12:
Addressed in https://github.com/infinition/Bjorn/pull/193 — you can override orientation in `config/shared_config.json` with `screen_reversed` / `web_screen_reversed` (see TROUBLESHOOTING.md).