infinition/Bjorn #114: [ERROR] An error occurred during: Installed Python requirements (Error code: 1)

nonobike, 2025-05-26:
```
[2025-05-26 11:25:14] [INFO] Updating E-Paper display configuration...
[2025-05-26 11:25:14] [SUCCESS] Updated E-Paper display configuration to epd2in13_V2
[2025-05-26 11:25:14] [INFO] Installing Python requirements...

Usage:   
  pip3 install [options] <requirement specifier> [package-index-options] ...
  pip3 install [options] -r <requirements file> [package-index-options] ...
  pip3 install [options] [-e] <vcs project url> ...
  pip3 install [options] [-e] <local project path> ...
  pip3 install [options] <archive url/path> ...

no such option: --break-system-packages
[2025-05-26 11:26:40] [ERROR] An error occurred during: Installed Python requirements (Error code: 1)
[2025-05-26 11:26:41] [ERROR] Check the log file for details: /var/log/bjorn_install/bjorn_install_20250526_111632.log
```

Hello,

When I use the installation script, I get the following error message.
Have you ever had this problem?

Arnaud

eimermusic, 2025-06-06:
I ran into the same thing today. Here is what I learned, diving down the rabbit hole.

Python packages can get corrupted or interrupted during download. If this happens, the local system will have this broken download cached and will try to compare any subsequent attempts at downloading and installing to the broken one.... and report the integrity hashes don't match.

This is what happened to me. Likely this is also what has happened to you.

I was not successful in purging caches or skipping caches. My connection to the package server was never reliable enough for one of the large packages (pandas). I had to step through the entire Bjorn install, manually.

Python-related steps are found in the install guide https://github.com/infinition/Bjorn/blob/main/INSTALL.md#step-3-bjorn-installation

Problem is, running that will likely just give you a better error message, not actually succeed. Here I ended up installing each package named in requirements.txt one by one. The failing package (pandas for me) I had to do the following for:

1. Try to install it, and get the error. But also this gives you the URL of the package file itself. It ends in a whl extension.
2. Download this file directly using wget.
3. Install this specific file.

Something like:

```
wget https://www.piwheels.org/simple/pandas/pandas-2.2.3-cp311-cp311-linux_armv6l.whl
sudo pip install pandas-2.2.3-cp311-cp311-linux_armv6l.whl
```

... and then move on the the next package... and the rest of the manual install steps.

DUptain1993, 2026-09-12:
Fixed in https://github.com/infinition/Bjorn/pull/193 — will close automatically when the PR is merged.